Ultimaker/Cura #11970: Random shorten walls/layers at the corner

536939, 2022-04-25:
### Application Version

5.0 Beta

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

I sliced an object. See image and project file.

### Actual results

Some walls are shorter in the corner than others.

### Expected results

I was expecting a normal corner.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

I scliced the same object whit 4.12.1, and it's work. Like the picture.

[image]

[CE3_dolphine.zip](https://github.com/Ultimaker/Cura/files/8558219/CE3_dolphine.zip)

nallath, 2022-06-15:
I've sliced this with the final 5.0.0 release and coudln't reproduce it. So it seems to have been fixed in the mean time.

closed 2022-06-15 by nallath

536939, 2022-06-17:
The error still exists.

[image]
[CE3_dolphine.zip](https://github.com/Ultimaker/Cura/files/8926399/CE3_dolphine.zip)

nallath, 2022-06-17:
This is caused by the maximum deviation setting. We have re-worked that for the 5.1 release.

As a work around you can play around with that setting to get the desired behavior.