Ultimaker/Cura #11971: Profile issues?

KnownOriginator, 2022-04-25:
### Application Version

5.0 beta

### Platform

Windows 10 latest update

### Printer

FLSun Super Racer

### Reproduction steps

Try to export profile.

### Actual results

Profile doesn't export.

### Expected results

Profile is exported.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I know it's a beta, and there *will* be issues such as this. I'm just hoping it's one of the things that's fixed for the next beta update.

KnownOriginator, 2022-04-25:
Side note, profiles made/updated in the current stable release (4.13.1 for me) are indeed updated/shown in the 5.0 beta, however they don't seem to go backwards. Profiles made on the 5.0 beta do not show in the 4.13.1 stable release for me.

sityware, 2022-04-26:
Same issue here, unable to export profiles.

nallath, 2022-04-26:
> Profiles made on the 5.0 beta do not show in the 4.13.1 stable release for me.

That is correct. This has never worked. Cura does not provide backwards compatibility for profiles.

Vandresc, 2022-04-28:
Devs, see CURA-9214 for our internal reference (Export of profile).

ronaldi40, 2022-05-05:
Habe das gleiche Problem in Cura 5 beta +1, Profile lassen sich definitiv nicht exportieren. Bitte um Lösungsvorschlag!
Have the same problem in Cura 5 beta +1, profiles definitely can't be exported. Please suggest a solution!

fvrmr, 2022-05-05:
Closing this issue, since it will be fixed for the 5.0 stable release!

closed 2022-05-05 by fvrmr