Ultimaker/Cura #11972: Cura 5.0.0-beta+1  - overwites existing gcode file even you select NO to overwrite

Yuccax, 2022-04-26:
### Application Version

5.0

### Platform

Windows 10

### Printer

other

### Reproduction steps

1. having sliced ready, click  "save to disk" 
2. select file that already exists
3. click "save"
4. click "No"
5. it is overwrite anyway

### Actual results

even it is selected do not overwrite. it is overwritten anyway

### Expected results

actin aborted 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

it doesn`t matter if SD card is used or using local disk

fvrmr, 2022-04-28:
Hi @Yuccax thank you for your report.
I can reproduce your issue, will bring it up with the team. Keep you posted.

MariskaMaas, 2022-04-29:
We discussed it with the team, this is unfortunate, we'll take a closer look. 
Thanks for reporting it! It can now be found CURA-9229

closed 2022-05-04 by nallath