Ultimaker/Cura #11973: v5b1 open 3mf file leads to inaccessible window

tispokes, 2022-04-26:
### Application Version

5

### Platform

W11

### Printer

-

### Reproduction steps

Ctrl+O, open a 3mf File. Now I can't see the primary/secondary button. 

### Actual results

[image]

After some switching between secondary and primary screen, I now see all options but can't click it.
This happens **only** when swithcing windows with shortcut Win+Shift+(Left | Right) as I use this very often.

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

not needed

fvrmr, 2022-04-28:
Hi @tispokes thank you for your report.
This is a known issue, we have a ticket on our backlog for this. Devs see CURA-9165
If you move the menu it will show the buttons.

tispokes, 2022-04-28:
> Hi @tispokes thank you for your report. This is a known issue, we have a ticket on our backlog for this. Devs see CURA-9165 If you move the menu it will show the buttons.

Yes i see the print-menu but cannot resize the settings menu. But as is it known bug, thanks for the information.

closed 2022-04-28 by tispokes