Ultimaker/Cura #11975: Excessive CPU usage when idle

Stutchbury, 2022-04-26:
### Application Version

5.0.0-beta+1

### Platform

Ubuntu 20.04

### Printer

Ender 3 Pro

### Reproduction steps

1. Download beta AppImage from here https://github.com/Ultimaker/Cura/releases/download/5.0-beta/Ultimaker-Cura-5.0.0-beta+1-linux.AppImage
2. Run 
3. Load model
4. Slice
5. Run top

### Actual results

CPU usage is continuously between 18-32%

### Expected results

CPU usage should be near zero when idle.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I have a warning on startup for two plugins: ProfilAnaliser and Octoprint - I have not deleted them because they are still in use on v4.13 and not available yet for v5.0.0

fieldOfView, 2022-04-26:
A prerelease of the OctoPrint plugin is available here:
https://community.ultimaker.com/topic/40313-fieldofview-plugins-for-cura-50/

nallath, 2022-04-26:
You can also delete them in the 5.0 configuration. You will not lose them for 4.13

fieldOfView, 2022-04-26:
Or Cura could have a "yeah I know, remind me when there is an update available and now stop nagging me for each plugin separately" checkbox. Please?

Stutchbury, 2022-04-26:
> You can also delete them in the 5.0 configuration. You will not lose them for 4.13

Thanks, have removed them and restarted - now get "There were some errors uninstalling packages" and a reminder to try 5.0.0-beta...

CPU still at ~20%

Stutchbury, 2022-04-26:
Uploading the cura.log because it now has more than 3 lines in it...

[cura.log](https://github.com/Ultimaker/Cura/files/8561957/cura.log)

fvrmr, 2022-05-02:
Hi @Stutchbury thank you for your report.
I observed the same today that the CPU is higher when idle. I will bring this up with the team. Keep you posted.

Stutchbury, 2022-05-02:
> Hi @Stutchbury thank you for your report. I observed the same today that the CPU is higher when idle. I will bring this up with the team. Keep you posted.

Thank you.

fvrmr, 2022-05-03:
I created a ticket for this on our backlog. 
Devs see CURA-9239

nallath, 2022-05-10:
This should be fixed for the 5.0 stable release.

closed 2022-05-10 by nallath