Ultimaker/Cura #11978: [5.0 b1] Unexpected "fill" in corners

Asterchades, 2022-04-26:
### Application Version

5.0.0 b1

### Platform

Windows 11 x64

### Printer

Custom FFF

### Reproduction steps

1. Created a custom model featuring additional geometry on top of a top surface, but smaller in size.
2. Slice said object

### Actual results

Depending upon how much of an "indent" is present between the previous layer and the next, some curious additional fill is added in the corners. In the model provided, sliced with the default Draft settings, this extra fill is visible on layers 469 and 470. If you switch to a 0.6mm line width but otherwise identical settings the same basic issue becomes evident on layers 98, 99, and 100 while 469 and 470 behave more in line with what's expected.

### Expected results

The expectation is that the inner walls themselves will occupy these extra areas, rather than being cut short and introducing additional (and unnecessary) movements.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Barbie Piano Leg R.zip](https://github.com/Ultimaker/Cura/files/8562787/Barbie.Piano.Leg.R.zip)

fvrmr, 2022-05-02:
Hi @Asterchades thank you for your report.
This is probably happening because your model is tapered. So in that case it is expected. 
Let me know what you think.

Asterchades, 2022-05-02:
It tapers, sure. But I've no idea why that would cause, or even justify, the presence of these additional skin lines - or that both impacted layers show different behaviour (469 only has fill in two corners while 470 has fill in all four). It would print both faster and cleaner if it behaved in the same way on those two layers (469 and 470) exactly the same as it did on the next two layers below that (467 and 468) which also make up the four layers of skin.

_Through experimentation I was also able to confirm that it's the taper on the part _above_ the impacted layers which is triggering the problem, not the taper on the impacted layers themselves._

What's more it also behaves more as expected when increasing the Wall Transitioning Angle to at least 24° from the default 10° (every value between 10° and 24° was either the same or worse, with 15° and 20° being examples of making it worse) or if the Skin Removal and Expansion values are both set to 0mm (0.4mm fixes the issues on layer 469 and 470, but introduces them on layer 100). Increasing the Wall Count to 3 (or setting the Wall Thickness to 1.2mm) also corrects the issue, as does increasing it to 4 (1.6mm) despite it preparing an almost identical line pattern to the two-wall version.

I'm also able to alter this behaviour by replacing the first layer of the upper-most taper in the base model with the same shape as should be presented by Slicing Tolerance: Middle. Theoretically this should have no effect at all on either the planner or the resulting printed shape, but it really doesn't like it. Notice that on this small section of the model (with the walls on the base straightened) it now extends down three layers (13, 14, and 15) instead of only two, with the first two affected layers only showing an issue in one corner.
[Test.zip](https://github.com/Ultimaker/Cura/files/8604031/Test.zip)

You can also alter the behaviour by changing the Top/Bottom Pattern, with Concentric behaving the worst (unexpected fill on all four layers). So it isn't actually fill but rather Skin being put into these unexpected voids.

nallath, 2022-05-23:
Bad bot!