Ultimaker/Cura #11985: Dual Extrusion where T0 is not used at all; CURA still sends gratuitous T0 and sets {initial_extruder_nr} to 0.

UberEclectic, 2022-04-26:
### Application Version

4.13.1

### Platform

macOS Monterrey

### Printer

Craftbot FLOW IDEX XL

### Reproduction steps

I have a project where I have two different materials loaded in a printer with dual extruders.
Specifically, I have PC in T0 and PLA in T1.
I am prototyping using the PLA in T1, until I get the part exactly right, and then I intend to switch to T0.
This is just me being lazy about changing filament, but it's a perfectly valid use-case for a dual extruder.

The problem is CURA is gratuitously sending a T0 command at the start of the job, and further, {initial_extruder_nr} is set to 0, not 1.

That's not supposed to happen.

This is my start g-code with the CURA tokens.
Everywhere you see {initial_extruder_nr} is supposed to be evaluated to 1 in this case, since T0 isn't used at all in the project.

M9006 S0                                         ;set DEFAULT mode 0=DEFAULT, 1=MIRROR, 2=PARALLEL, 3=not used, 4=INVERTHEAD, 5=BACKUP
                                                 ;if you want to be able to set the mode on the HMI, you have to comment out the above line
M900 K0                                          ;turn off linear advance
G28 X Y                                          ;home X and Y axes
G28 Z

M104 T{initial_extruder_nr} S{material_standby_temperature, 0}       ;set extruder temp (no wait)
M140 S{material_bed_temperature_layer_0}         ;set bed temp (no wait)

G21                                              ;mm
G90                                              ;absolute positioning
G92 E0                                           ;reset extruder

M425 S1                                          ;use wiper blade
M430 Z1 F1800                                    ; Set head change elevation 1mm
M431 E4 F1800                                    ;Head change exit retract 4mm
M433 E4 F1800                                    ;Head change enter prime 4mm

                                                 ;wait for all temperatures
M190 S{material_bed_temperature_layer_0}         ;set bed temp (wait)
M109 T{initial_extruder_nr} S{material_print_temperature_layer_0, 0} ;set extruder temp (wait)

;move head to the wipe position so any ooze ends up in the bucket
G1 F1800 Y160

M300                                             ;signal start of print
T{initial_extruder_nr}                           ;Send gratuitous T0, T1 (avoids defaulting to T0).
;Pre-extrude some filament. That way the nozzle is primed and when the print starts,
;the filament is wiped off.
;G1 F1800 E5

### Actual results

This is what CURA generated.  Notice the gratuitous T0 and later on where it should have put T1 it put T0 again.

;Generated with Cura_SteamEngine 4.13.1
T0
M82 ;absolute extrusion mode
M9006 S0                                         ;set DEFAULT mode 0=DEFAULT, 1=MIRROR, 2=PARALLEL, 3=not used, 4=INVERTHEAD, 5=BACKUP
;if you want to be able to set the mode on the HMI, you have to comment out the above line
M900 K0                                          ;turn off linear advance
G28 X Y                                          ;home X and Y axes
G28 Z

M104 T0 S160.0       ;set extruder temp (no wait)
M140 S90         ;set bed temp (no wait)

G21                                              ;mm
G90                                              ;absolute positioning
G92 E0                                           ;reset extruder

M425 S1                                          ;use wiper blade
M430 Z1 F1800                                    ; Set head change elevation 1mm
M431 E4 F1800                                    ;Head change exit retract 4mm
M433 E4 F1800                                    ;Head change enter prime 4mm

;wait for all temperatures
M190 S90         ;set bed temp (wait)
M109 T0 S260.0 ;set extruder temp (wait)

;move head to the wipe position so any ooze ends up in the bucket
G1 F1800 Y160

M300                                             ;signal start of print
T0                           ;Send gratuitous T0, T1 (avoids defaulting to T0).
;Pre-extrude some filament. That way the nozzle is primed and when the print starts,
;the filament is wiped off.
;G1 F1800 E5
M83 ;relative extrusion mode
G1 F1200 E-1
;LAYER_COUNT:599
;LAYER:0
M107

### Expected results

There should be no gratuitous T1 and {initial_extruder_nr} should evaluate to 1.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I should point out that even disabling T0 is fraught with issues where only T1 is used in the project.
If I disable T0, then beyond the scenes bed adhesion and support get changed to T0 and are then hidden and I cannot change them back to T1.
That is why I had to leave T0 enabled.
However as T0 is not actually used, I have stepped into the issue that is the subject of this bug.

UberEclectic, 2022-04-26:
The project: [CFIXL_30mm Disc 0.25mm Thick 3MF.zip](https://github.com/Ultimaker/Cura/files/8564570/CFIXL_30mm.Disc.0.25mm.Thick.3MF.zip)

The generated G-Code: [CFIXL_30mm Disc 0.25mm Thick G-CODE.zip](https://github.com/Ultimaker/Cura/files/8564573/CFIXL_30mm.Disc.0.25mm.Thick.G-CODE.zip)

GregValiant, 2022-04-27:
That 3mf is a model file.  Use the "File | Save Project" command and zip and post that 3mf.

UberEclectic, 2022-04-27:
@GregValiant I used COMMAND-S (on Mac).  Not the same thing?  It always works for me and when I load such a 3MF I am prompted as to whether to just load the models or treat it as a project.  I'll do it manually from the menu, shortly, just in case.

GregValiant, 2022-04-27:
I don't know Mac's.  Maybe Cmd-S is Export instead of Save Project?
When I opened that previous 3mf file my printer stayed active.  No "dual extruder" and your settings didn't come in.

UberEclectic, 2022-04-27:
Cmd-S is save project.
[image]

But I'll try again...

UberEclectic, 2022-04-27:
Okay, I re-created the project from scratch and saved it from the menu.  Should be no different than Cmd-S.

[CFIXL_Canyon Light GoPro Mount 3MF.3mf.zip](https://github.com/Ultimaker/Cura/files/8572757/CFIXL_Canyon.Light.GoPro.Mount.3MF.3mf.zip)

[CFIXL_30mm Disc 0.25mm Thick G-CODE.gcode.zip](https://github.com/Ultimaker/Cura/files/8572760/CFIXL_30mm.Disc.0.25mm.Thick.G-CODE.gcode.zip)

UberEclectic, 2022-04-27:
@GregValiant If you're still having problems with my uploads, this is super easy to reproduce.

1) Duplicate any dual extruder printer you may have
2) Replace the start G-Code with mine from the original problem statement
3) Keep both T0 and T1 active.
4) Load any model you wish
5) Set the model to print from T1

You'll see the bug when you inspect the G-Code produced.

For the bonus bug, try disabling T0, and see if you can actually get T1 to be allowed for use in bed adhesion, support, etc.

UberEclectic, 2022-04-27:
Actually, I did just manage to workaround the problem re: the so-called "bonus bug" ...
I had to find all the hidden extruder settings that were set to "Not overridden" and explicitly bind them to T1.
Then I could successfully disable T0.

UberEclectic, 2022-04-27:
@GregValiant I think I spotted the original issue.
By searching for "extruder" in the settings, I can see that Build Plate Adhesion is set in the profile to either Left or Right, and there isn't a "Not overridden" option like all the others.
So, really, I should have set the bed adhesion extruder to Right Extruder (T1).
I did just confirm that if I take care to unhide all extruder related settings (which could be hidden depending on support and adhesion options), then I can get the expected behavior re:  {initial_extruder_nr}.

The confusion here is caused by dual extrusion settings that may potentially be hidden.
It would help if when disabling T0, that all extruder settings (including hidden) would get shifted to T1 automatically.

GregValiant, 2022-04-27:
You aren't the first this has happened to.  If you have T0 disabled and use a profile that calls it then problems occur.
Another thing that happens is with support blockers.  They always come in assigned to T0.  That makes sense as all printers have a T0 but it's another thing that maybe could be handled better.

ladismrkolj, 2022-05-04:
Hey I have the same problem as above, but disabling T0 extruder (with the checkbox) solves the problem. However in a dual color print T0 is always initial_extruder_nr, regardless if the model actually begins with T1.