Ultimaker/Cura #11988: Switching machines adds extra Extruders to Settings menu, then crashes when the mouse-hovers over them.

pufferfish-tech, 2022-04-26:
### Application Version

5.0 beta

### Platform

Windows 10

### Printer

Any (Ender, FLSUN, Artillery)

### Reproduction steps

1. Change printer using the combo box below the menu (not the menu). 
2. You can do this as many times as you want (each time will add another instance in the menu in step 4)
3. click Settings
4. Observe the extra menu items labeled "Extruder 1"
5. If you press escape or the menu top level here, cura wont crash, but hovering any of these extruder menu items causes instant disappear of cura UI. 

### Actual results

Cura disappears. 

### Expected results

Cura should not disappear. More specifically those menu items should not be present. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

There does not appear to be an entry at all in the log file, however I believe this is easy enough to reproduce on any setup with more than one printer.
Image should show just one "Extruder 1"
[image]
 
Additional trivia, if selecting a dual extrusion printer, you get a bonus "extruder 2". So it looks like it's just adding the new ones, but keeping the old ones.
[image]

pufferfish-tech, 2022-04-26:
I should add that the workaround is to use the button pictured and avoid the menu entirely, then it does not appear to crash.
[image]

jellespijker, 2022-04-26:
Thank you for reporting this.
Keep up the nice videos we really enjoy them.

Sonicboom247, 2022-04-27:
Happening to me as well, seems that continuing to switch between machines keeps adding extruders 
[image]

Hun73rdk, 2022-04-28:
Can confirm happens to me also had 8 extruders last time

fvrmr, 2022-04-28:
Hi @pufferfish-tech thank you for your report.
I can reproduce your issue, will bring it up with the team. Keep you posted!

MariskaMaas, 2022-04-28:
Thanks for reporting this! 
For internal reference, this is CURA-9222

nallath, 2022-05-02:
Fixed for the final 5.0 release. Thanks for reporting it!

closed 2022-05-02 by nallath