Ultimaker/Cura #11989: Line overlaid on existing line in same layer

DaveLeddon, 2022-04-26:
### Application Version

4.13.1

### Platform

Windows 10

### Printer

Voron 2.4 300mm

### Reproduction steps

Printed a rectangle two layers thick with several round holes near the center
[cura.log](https://github.com/Ultimaker/Cura/files/8566533/cura.log)
[V300_Dave Enclosure Front.zip](https://github.com/Ultimaker/Cura/files/8566535/V300_Dave.Enclosure.Front.zip)
.

### Actual results

Cura prints lines until encountering a hole at which point it picks another location to continue printing lines.  At some point it returns to the hole and prints lines back to the point where it left off but seems to print an extra line on top the last existing line.  It does this in both layers and results in ridges in the finished result.

### Expected results

It would be perfect if not for the overlaid lines.  I tried using Cura 5.0.0 but still got ridges.  This may be the result of a Cura setting that I don't understand.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[V300_Dave Enclosure Front.zip](https://github.com/Ultimaker/Cura/files/8566552/V300_Dave.Enclosure.Front.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/8566554/cura.log)

GregValiant, 2022-04-27:
I doubt that the problem is overlaid lines.  The normal situation is what you saw.  The nozzle prints say from left to right, encounters an obstacle to it's current path, continues past the obstacle leaving a bare spot.  When it comes back to finish the bare spot it might print it from right to left.  If you have a bit of over-extrusion then the nozzle is "pushing a wave".  It left a slight ridge when moving left to right and then when it came back it left another slight ridge moving right to left.  The 2 waves end up on top of one another leaving a ridge (a moraine?).
The fix is "Monotonic Top/Bottom Order".  All the moves on a skin will be in the same direction and so the ridge keeps moving until it gets pushed to where the skin ends.  Monotonic can make for some extensive travel moves but it will leave a skin looking much better and without the ridges caused by direction changes.
Another thing you can do is to enable Ironing which is time intensive but does what it says.  I found that with it enabled and the Ironing Flow at 7% the finish was excellent.  Experimenting will tell you.  If the Ironing Flow is too high you get a huge ridge.

DaveLeddon, 2022-04-27:
Hello Greg,
 
I made two test prints, one using my original settings that generates ridges about one line width wide:
 
And a second with Monotonic Top/Bottom Order and Ironing turn on which now generates trenches about three line widths wide:
 
Ironing did tend to smooth over the trenches somewhat but the underlying problem is still visible.
 
Any ideas?

-- 
Best regards,
Dave Leddon

GregValiant, 2022-04-28:
Post one of those gcodes and also a project file (bring in the model, get Cura ready to slice with your settings, use the "File | Save Project" command to generate a 3mf project file.)  Zip the gcode and the project file and post the Zip file here.

DaveLeddon, 2022-10-11:
Hello Greg,
 
Interesting how the anomaly went from ridge to rut by turning on Monotonic Top/Bottom Order.

-- 
Best regards,
Dave Leddon