Ultimaker/Cura #11990: Cura 5 Object move dialog is unusable when there are many extruders defined.

Anathema-Device, 2022-04-26:
### Application Version

5.0.0.beta-1

### Platform

Windows 10

### Printer

Custom Geeetech A10M

### Reproduction steps

Create a printer definition with 8 extruders
Open any .stl file
Select the object on the platter
Select the move icon (top left)
Try and enter coordinates < the entry fields for the coordinates are behind the top icon bar that shows the folder, printer name and extruders
This issue also exists in Cura 4.13.1 and earlier.

[image]

### Actual results

It is not possible to accurately adjust the position of objects on the build plate when this happens.

### Expected results

The dialog should be positioned lower so that the values can be edited.  

One simple solution would be to put the column of extruders to the left of the column of icons with the move, rotate, etc icons when they get too long?  This would also solve the issue of not being able to get to all the extruder icons when the window is too small.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional info, I believe the steps above will show the problem clearly - also see the picture attached

fvrmr, 2022-04-28:
Hi @Anathema-Device thank you for your report.
I will bring this up with the team, keep you posted.

fvrmr, 2022-05-02:
We have discussed it with the team and decided to defer this issue, this is not an issue for Ultimaker printers and that's why it will not get a high priority for us to fix. 
Anyone is welcome to fix this and create a pull request for this.

fieldOfView, 2022-05-02:
A quick workaround is installing the prerelease of the SidebarGUI plugin, which does not have this issue.
https://community.ultimaker.com/topic/40313-fieldofview-plugins-for-cura-50/

I will see if I can create a fix for the standard GUI as well.

fieldOfView, 2022-05-02:
See my fix in #12054

PR #12054 (2022-05-18): Fix toolbar positioning when the toolbar is taller than its parent
This PR fixes the z-order of toolpanels and the placement of the toolbar when the toolbar becomes longer than the window permits (eg through multiple additional tool-plugins and/or many extruders).

Fixes #11990
closed 2022-05-18 by nallath