Ultimaker/Cura #11991: Save to disk not working

jtradermli, 2022-04-26:
### Application Version

5.0.0

### Platform

linux penquin

### Printer

ender3

### Reproduction steps

installed cura 5.0.0
opened cura 5.0.0
loaded an stl file
changed filament temp to 217, bed temp to 60, infill to 30, and removed supports
sliced model
when selecting Save to Disk, nothing happened
opened cura 4.13, worked 
went back to cura 5.0, still nothing when attempting to Save to Disk
did this 3 times
deleted cura 4.13
opened cura 5.0.0 and still the same problem

### Actual results

nothing

### Expected results

save .gcode to file

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2022-04-28:
Hi @jtradermli thank you for your report.
Could you share your log file? You can find your log file here: `$HOME/.local/share/cura/<Cura version>/cura.log`

jszymon, 2022-04-30:
Same here, but Cura freezes on save to disk (keeps using the processor).  Log file attached:

[cura.log](https://github.com/Ultimaker/Cura/files/8597810/cura.log)

mjrider, 2022-05-02:
i had a similar issue yesterday, app seems to freeze, and nothing happening.
What did happen that the window of the save file dialog was opened, but the window wasn't raised or focused, hiding it behind the cura window, after focusing that window and saving the file, everything worked as expected

JalonDodson, 2022-05-14:
I'm having this issue as well. 
@mjrider, unfortunately that didn't work for me. My Cura logs don't indicate any sort of errors. Not sure what's happening. I will say that it doesn't freeze when I press 'save to disk', like @jszymon, but it just doesn't work. 

Might be worth noting: Unable to open any file browsing windows within Cura 5.0.0.

AonoZan, 2022-06-09:
This issue still persist. I don't get any file dialog when I click on "Open file(s)", "Save Project" or "Save to file" (when model is sliced). File dialog window never shows up. I'm on Xubuntu and this is only 5.0.0 version issue, 4.12 worked fine no issues.

Nakamura-shot, 2022-06-25:
Same for me with Cura 5.0. Save dialog box just hangs and then crashes as unresponsive. Went back to 4.13 and the save box also hangs.

brianinmaine, 2022-06-30:
arch linux - xfce - cura 5 appimage. save to disk only works when a portable drive is mounted, then it says save to portable disk and it works. when just hitting save to disk nothing happens.

ggwpez, 2022-08-24:
Same on Ubuntu 22.04 with V5.1

Putting in an USB drive makes it work with the "Save to Removable Drive" dialog. Otherwise nothing happens.

RamiGspo, 2022-09-23:
Same issue on Ubuntu 22.04 with 5.2.0-alpha

Putting in an USB drive makes it work with the "Save to Removable Drive" dialog. Otherwise nothing happens.

MariMakes, 2022-10-05:
This issue was closed automatically, but I'm opening it again since it's clearly still present. 
I'll bring it up to the attention of the team again 🤞 

In the meantime. Logs are very welcome to help troubleshoot. 👼 
You can find your log file here:
Ubuntu/Linux: $USER/.local/share/cura/<Cura version>/cura.log

MariMakes, 2022-11-01:
We've also had this report which also seems to be related to the system-provided file dialog: https://github.com/Ultimaker/Cura/issues/12659

CocaineJohnsson, 2022-12-23:
[cura.log](https://github.com/Ultimaker/Cura/files/10295559/cura.log)

Save to disk isn't working, save project isn't working.

Cura 5.2.1 -- arch linux (zen kernel 6.0.12), bspwm

scabux, 2023-01-28:
> first i thought to have the same issue using 5.2.1 linux-modern AppImage on arch linux.
> 
> however, it turns out to work with the aur package `aur/cura-bin 5.2.1-4`

For me, this version does not work either. Same problem for the "open project" button, also if I try it via the menu. Logs do not seem to show anything relevant.

iluvatyr, 2023-02-21:
This issue is same for me. Cannot save to disk with any 5.x version, whereas save to removable media works.  with version 4.13 it works both ways.

mwri, 2023-02-24:
I can't save to disk either, I don't get the CPU usage reported by some, I click "save to disk" and it just does nothing, and logs nothing. I have attached a complete log in case it helps though, from start up, slicing something, and trying to save.

It does persist in logging this stuff a lot:
```
2023-02-24` 09:13:24,086 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 76800 works. Retry nr: 0. Wait timeout: 3
2023-02-24 09:13:28,600 - DEBUG - [JobQueueWorker [0]] USBPrinting.AutoDetectBaudJob.run [49]: Checking /dev/ttyUSB0 if baud rate 57600 works. Retry nr: 0. Wait timeout: 3
```

There is a /dev/ttyUSB0, but it is not a 3D printer.

I also experience #12266 (the problem that I can't open an STL file).

Saving to removable storage does work, but I don't want to do that.

[cura.log](https://github.com/Ultimaker/Cura/files/10822541/cura.log)

kuene, 2023-02-24:
Same Problem. Can not open or save Files. No dialog window appears.

kuene, 2023-02-24:
apt install xdg-desktop-portal-gtk

fixed the Problem for me

iluvatyr, 2023-02-24:
> `apt install xdg-desktop-portal-gtk`
> 
> fixed the Problem for me

true, for me too

AonoZan, 2023-02-24:
Newest Cura 5.2.2 works just fine without any additional installation for me.

Alex20129, 2023-02-25:
OS: ``Ubuntu 20.04.5 LTS (Focal Fossa)``
APP: ``Cura 5.3.0-beta.2``
"Save to Disk" doesn't work
"File" -> "Open file(s)" doesn't work
"File" -> "Export" doesn't work

Alex20129, 2023-02-25:
after i did
``apt install xdg-desktop-portal-gtk``
all dialogs showing up properly, all works just fine

scabux, 2023-03-04:
On Arch Linux, installing `xdg-desktop-portal-gtk` was not sufficient. However, running Cura with the following options solved the issue for me:
`cura -platformtheme gtk3`

MariMakes, 2023-03-31:
Quick update on our side. 

We might have a fix for this issue. 🎉 
It's a bit hard for us to test, but we have a nightly build available. 
Can you please try it out and let us know if it resolves your issue? 

[You can find the download links and the discussion here. ](https://github.com/Ultimaker/Cura/discussions/15098)

mennucc, 2023-03-31:
First and foremost, I am experiencing a different problem. 
  I click "save" then the save dialog appears, when I type any letter into the widget to specify the save name, the save dialog jumps into "search mode" and then it is almost impossible to specify the save name. 
This problem appears in every version of Cura. I use Ubuntu 22.04 (with Mate desktop) . This said , adding the option  _-platformtheme gtk3_ solves the problem

closed 2023-10-19 by MariMakes