Ultimaker/Cura #11995: crashing

glennwad, 2022-04-27:
### Application Version

5

### Platform

windows 8.1

### Printer

ender3v2

### Reproduction steps

after i select stl it puts it on the build plate and as soon as i click on part it crashes if i slice it with out doing anything to the stl its fine

### Actual results

program shuts down

### Expected results

i should be able to resize or move stl

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

no

glennwad, 2022-04-28:
> Please provide log files, we can't help you without them.

i couldnt locate the log
ill try again

glennwad, 2022-04-28:
is this the log you need

fieldOfView, 2022-04-28:
Unfortunately, github strips out attachments from email replies. Can you please use the github web interface to add the logs?

glennwad, 2022-04-29:
no i just uninstalled 5 ill wait till its final is out

closed 2022-04-29 by glennwad