Ultimaker/Cura #11998: Cura 5.0 Beta crashes when starting in Linux Mint 20.3 unless started with root access

mttr402, 2022-04-27:
### Application Version

5.0 beta

### Platform

linux mint

### Printer

Ender 6

### Reproduction steps

Running with appimageLauncher to lauch Cura

### Actual results

Traceback (most recent call last):
  File "cura_app.py", line 238, in <module>
  File "cura/CuraApplication.py", line 813, in run
    super().run()
  File "UM/Qt/QtApplication.py", line 366, in run
    super().run()
  File "UM/Application.py", line 256, in run
    self.initialize()
  File "cura/CuraApplication.py", line 323, in initialize
    self.__sendCommandToSingleInstance()
  File "cura/CuraApplication.py", line 339, in __sendCommandToSingleInstance
    if self._single_instance.startClient():
  File "cura/SingleInstance.py", line 32, in startClient
    if single_instance_socket.state() != QLocalSocket.ConnectedState:
AttributeError: type object 'QLocalSocket' has no attribute 'ConnectedState'

### Expected results

to start normally

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Thread 0x00007fe4d1ffb700 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00007fe4d27fc700 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00007fe4d2ffd700 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00007fe4d37fe700 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00007fe4d3fff700 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00007fe4e8845700 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00007fe4e9046700 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Thread 0x00007fe4e9847700 (most recent call first):
  File "threading.py", line 320 in wait
  File "threading.py", line 460 in acquire
  File "UM/JobQueue.py", line 98 in _nextJob
  File "UM/JobQueue.py", line 124 in run
  File "threading.py", line 1009 in _bootstrap_inner
  File "threading.py", line 966 in _bootstrap

Current thread 0x00007fe52880db80 (most recent call first):
  File "cura/CrashHandler.py", line 396 in _logInfoWidget
  File "cura/CrashHandler.py", line 182 in _createDialog
  File "cura/CrashHandler.py", line 86 in __init__
  File "cura_app.py", line 176 in exceptHook

peteruithoven, 2022-04-30:
Same on elementary OS 6.1 Jólnir (Built on Ubuntu 20.04.3 LTS). 
I tried renaming the 5.0 config file (`/home/peteruithoven/.config/cura/5.0/cura.cfg`) but that does not help.

jeferod83, 2022-06-18:
I've a problem using Cura 5 AppImage, but I solved. In my case, Cura starts, but it shows me a lot of erros related to plugins that cannot be uninstalled, and if I try to remove another error grows up.

But, in my case, the problem is with folder permissions. Doing a `chown` in the cura files, located at `~/.local/share/cura/`, solved all my problems related to Cura startup.

peteruithoven, 2022-07-20:
> But, in my case, the problem is with folder permissions. Doing a chown in the cura files, located at ~/.local/share/cura/, solved all my problems related to Cura startup.

Could you clarify? You changed all files there to your own user?

peteruithoven, 2022-07-20:
This actually worked for me: https://github.com/Ultimaker/Cura/issues/12343#issuecomment-1161683871

MariMakes, 2023-10-19:
👋 Quick update on our side. 

This is resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-19 by MariMakes