Ultimaker/Cura #12000: (5.0.0) macOS Unhandled Exception on launch: dlopen ... Symbol not found: _ffi_closure_alloc

jwigum, 2022-04-27:
### Application Version

5.0.0

### Platform

macOS 10.14.6 (18G9323)

### Printer

Custom

### Reproduction steps

Attempt to launch Cura 5.0.0 after install. Application verifies, and then prompts to open. Choose to open, dock icon bounces, then stops.

### Actual results

Console prints the following for process Ultimaker Cura: 

Failed to execute script 'cura_app' due to unhandled exception: dlopen(/Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so, 2): Symbol not found: _ffi_closure_alloc
  Referenced from: /Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libffi.dylib
 in /Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so

### Expected results

Cura should have launched.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I don't have a log file for 5.0 (only for 4.12 and 4.13). 

4.13.1 launches without issue.

fvrmr, 2022-05-02:
Hi @jwigum thank you for your report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/12001, so I will close this issue.

closed 2022-05-02 by fvrmr

jwigum, 2022-05-02:
Why is this being closed as a duplicate of a later report? Literally 1200 closed for 1201.

jwigum, 2022-05-02:
@fvrmr Can this be opened and used as the tracked bug? The other seems very generic.