Ultimaker/Cura #12001: MacOS dmg will install but not open. It gives prompt but does not boot.

CryptoTheDragon, 2022-04-27:
### Application Version

5.0.0 beta

### Platform

macOS Mojave 

### Printer

ender 3 pro

### Reproduction steps

just try it on macOS Mojave 

### Actual results

it does not boot up

### Expected results

it should boot up and run normally.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Yeah it broken for Mac

zanardozanardo, 2022-04-28:
Same problem here. 
MacOS Mojave 10.14.6. 
Doesn't open the Cura 5.0.0 beta.
=(

bitnautist, 2022-05-01:
Yup, same problem here on my MacOS Mojave 10.14.6 and Mac Pro Dual 6 core with 128GB ram

fvrmr, 2022-05-02:
Hi @CryptoTheDragon thank you for your report.
Theoretically we support it, but we didn't test this on 10.14.6. We will check our selves if we can reproduce this. Keep you posted.

fvrmr, 2022-05-05:
We did some investigation for this issue and it turns out that unfortunately we can't support 10.14.6 anymore because we switched to a package manager that only supports Catalina and above. And since Mojave is also not official supported anymore we don't plan to fix this.

closed 2022-05-05 by fvrmr

Paraphraser, 2022-05-06:
> switched to a package manager that only supports Catalina and above

Could I please urge you to at least consider:

1. Setting the relevant Info.plist keys ([`LSMinimumSystemVersion`](https://developer.apple.com/documentation/bundleresources/information_property_list/lsminimumsystemversion) and/or [`MinimumOSVersion`](https://developer.apple.com/documentation/bundleresources/information_property_list/minimumosversion)) for Cura 5 so that macOS will give users running Mojave and earlier appropriate feedback; and
2. Documenting "Catalina and later" in some reasonably prominent fashion, such as in the list of [Cura dependencies](https://github.com/Ultimaker/Cura#dependencies).

<hr> 

For anyone finding this issue and wondering about the "single Dock bounce, no crash dialog, no log directory" behaviour on Mojave:

``` 
$ /Applications/Ultimaker-Cura.app/Contents/MacOS/Ultimaker-Cura
Traceback (most recent call last):
  File "cura_app.py", line 27, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "cura/CrashHandler.py", line 27, in <module>
    from UM.Application import Application
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Application.py", line 9, in <module>
    from UM.Controller import Controller
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Controller.py", line 3, in <module>
    from UM.Scene.Iterator.DepthFirstIterator import DepthFirstIterator
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Scene/Iterator/DepthFirstIterator.py", line 3, in <module>
    from UM.Scene.SceneNode import SceneNode
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Scene/SceneNode.py", line 14, in <module>
    from UM.Mesh.MeshBuilder import MeshBuilder
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Mesh/MeshBuilder.py", line 4, in <module>
    from UM.Mesh.MeshData import MeshData
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Mesh/MeshData.py", line 16, in <module>
    import scipy.spatial
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "scipy/__init__.py", line 153, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "scipy/_lib/_ccallback.py", line 1, in <module>
  File "_ccallback_c.pyx", line 210, in init scipy._lib._ccallback_c
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "ctypes/__init__.py", line 8, in <module>
ImportError: dlopen(/Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so, 2): Symbol not found: _ffi_closure_alloc
  Referenced from: /Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libffi.dylib
 in /Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so
[820] Failed to execute script 'cura_app' due to unhandled exception: dlopen(/Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so, 2): Symbol not found: _ffi_closure_alloc
  Referenced from: /Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libffi.dylib
 in /Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so
[820] Traceback:
Traceback (most recent call last):
  File "cura_app.py", line 27, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "cura/CrashHandler.py", line 27, in <module>
    from UM.Application import Application
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Application.py", line 9, in <module>
    from UM.Controller import Controller
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Controller.py", line 3, in <module>
    from UM.Scene.Iterator.DepthFirstIterator import DepthFirstIterator
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Scene/Iterator/DepthFirstIterator.py", line 3, in <module>
    from UM.Scene.SceneNode import SceneNode
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Scene/SceneNode.py", line 14, in <module>
    from UM.Mesh.MeshBuilder import MeshBuilder
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Mesh/MeshBuilder.py", line 4, in <module>
    from UM.Mesh.MeshData import MeshData
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "UM/Mesh/MeshData.py", line 16, in <module>
    import scipy.spatial
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "scipy/__init__.py", line 153, in <module>
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "scipy/_lib/_ccallback.py", line 1, in <module>
  File "_ccallback_c.pyx", line 210, in init scipy._lib._ccallback_c
  File "PyInstaller/loader/pyimod03_importers.py", line 495, in exec_module
  File "ctypes/__init__.py", line 8, in <module>
ImportError: dlopen(/Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so, 2): Symbol not found: _ffi_closure_alloc
  Referenced from: /Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so (which was built for Mac OS X 10.15)
  Expected in: /usr/lib/libffi.dylib
 in /Applications/Ultimaker-Cura.app/Contents/MacOS/lib-dynload/_ctypes.cpython-310-darwin.so
```

Cura 5.0.0-beta+1 does work on macOS 11.6.5 (Big Sur) and macOS 10.15.7 (Catalina).