Ultimaker/Cura #12005: Errore DLL 1114

2G3D, 2022-04-28:
### Application Version

5 beta+1

### Platform

win 10 pro 64bit

### Printer

Geeetech A10

### Reproduction steps

open program Cura

### Actual results

[image]

### Expected results

program not opened

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Radeon HD7400M notebook HP Pavilion G6 8GB RAM

DoganM95, 2022-04-30:
Try a reinstall. Looks like just a missing dll (failed installation).

fvrmr, 2022-05-02:
Hi @DoganM95 thank you for your report.
Could you try to reinstall like @DoganM95 suggested, let us know if that works for you.

closed 2022-05-23 by github-actions[bot]

mMerlin, 2023-12-16:
For others landing here from a search. I had the same error on Windows 10 with version 5.2.1 and 5.6.0. Both installed fine, but failed on first (and every) run. Reinstall did not help.

Summary: It's a graphics software problem.
FIX3 from
https://www.minitool.com/data-recovery/how-to-start-windows-10-in-safe-mode-009.html
worked for me. Using AMD Catalyst Control Center instead of NVIDA Control Panel. Change the power setting to maximum performance for ultimaker-cura.exe.

Warning: FIX1 from that page made my laptop unbootable. I had to go through safe mode to undo the global change. Which only undid one of the 2. I had to do it again the 'nomal' way after another reboot.