Ultimaker/Cura #12006: Unexpected travels in Vase mode causing unnecessary seam

Ferrograph, 2022-04-28:
### Application Version

5.0.0-beta

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Slice the attached project with included profiles. Look at the travels on the bottom layer
[cura.log](https://github.com/Ultimaker/Cura/files/8581760/cura.log)
.

### Actual results

Slicing produces weird unnecessary travels as shown in the screen shot resulting in ugly seams. 

[image]
[image]

### Expected results

Slicing in 4.13.1 does not have unnecessary travels and expected single seam.
[Cura Belt Project.zip](https://github.com/Ultimaker/Cura/files/8581630/Cura.Belt.Project.zip)
.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Im trying to print belts with minimal seam and uniform cross section.

fvrmr, 2022-05-04:
Hi @Ferrograph thank you for your report.
I will discuss this with the team. Keep you posted.

fvrmr, 2022-05-05:
I have discussed it with the team and we decided to defer this issue for now. It will not get a high priority to fix for us now. I hope you understand.

closed 2023-06-09 by github-actions[bot]

Ferrograph, 2025-04-18:
Wondering if this will ever be fixed? Because of this issue I cant use Cura for my prints