Ultimaker/Cura #12012: Model prints incorrectly with new beta version

SwagDad99, 2022-04-29:
### Application Version

5.0.0 Beta+1

### Platform

Windows 11

### Printer

Creality CP-01 AND JGMaker Artist-D Pro

### Reproduction steps

Attempt to print a model that printed well on 4.13.1

### Actual results

The tines of the comb printed correctly with the old version.  With the new version, they print "double", and are not secured to the base correctly.  The same stl was loaded into each version of Cura, with the same settings, and they printed differently.  I have included a pic showing both combs.  The one on the left is correct and printed with 4.13.1.  The one on the right is printed with the new Beta 5.0.0 version.  Identical settings, and same stl file.
.2 mm layer

[image]

### Expected results

It should print the same, or better with the new version of Cura.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I actually spent quite a bit of time and frustration trying to figure out what I was doing wrong, until I realized it was the new slicer that was causing the grief.

GregValiant, 2022-04-29:
Load the model in Cura 5.0B and set it up to slice.  Use the File | Save Project command then zip the 3mf file and post it here.  Someone will need to take a look at it to figure out what's going on.

SwagDad99, 2022-04-29:
> 

Where does it save the project file?  I don't get a dialog box or anything...
Uh oh...  When I go to preferences, all I see is an empty box with the preference 'types' listed on the left, (General, Settings, Printers, Materials and Profiles), and a large empty box.  Nothing when I choose one from the list..
[image]
.

SwagDad99, 2022-04-29:
Restarted everything, and was able to save the project.  Here is the zi
[CE5_Folding Beard Comb (Blade v10).zip](https://github.com/Ultimaker/Cura/files/8588150/CE5_Folding.Beard.Comb.Blade.v10.zip)
p

GregValiant, 2022-04-29:
Because each of the tines of the comb are at an angle, there are steps.  At the end of each step there is a travel move to the adjacent tine.  4.13 makes a path with two outer walls (one up the tine and the other back.  The default settings in 5.0 makes a single pass up a tine, moves laterally to the next tine, and makes a single pass coming back.  Oozing during that lateral move is the problem.
You might be able to make an adjustment to "Split middle line threshold" to 50% and see how it looks.  It's a new setting and doesn't have an equivalent in 4.13.