Ultimaker/Cura #12015: Cura 5 takes a very long time to slice specific objects.

Susanoo1337, 2022-04-29:
### Application Version

5.0.0-beta+1

### Platform

Windows 10

### Printer

Ender 5

### Reproduction steps

1. Load model
2. Press Slice
3. Wait for 212 seconds

### Actual results

Slicing bar fills halfway, then stops, after ~200 seconds it resumes.

### Expected results

Most objects, even larger and more complex objects with supports, slice in around 3-20 seconds.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura - Copy (2).log](https://github.com/Ultimaker/Cura/files/8589275/cura.-.Copy.2.log)
[CE5_Wrench.zip](https://github.com/Ultimaker/Cura/files/8589277/CE5_Wrench.zip)

fvrmr, 2022-05-02:
Hi @Susanoo1337 thank you for your report. 
I can reproduce this with your model, I will bring it up with the team. Keep you posted.

fvrmr, 2022-05-03:
I have discussed it with the team and we decided for now to defer this issue. Since it will eventually slice, we know it is a bit annoying to wait that long. But since it is only happening with one model it will not get a high priority for us to fix.
If this happens with more models and other users we will discuss this again. I hope you understand.

closed 2023-06-09 by github-actions[bot]