Ultimaker/Cura #12016: Issue with support if "Use Adaptive Layer" is enabled

Skybeat, 2022-04-29:
### Application Version

5.0 beta

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

Slice a object what needs support with "Use Adaptive Layer" disabled and support enabled
[image]

Slice the same object  with "Use Adaptive Layer" enabled and support enabled
[image]

As you can see there is support missing as soon as "Use Adaptive Layer" is enabled and the object is sliced. 
In the case of the images, you and see that with Use Adaptive Layers, Cura try to print a part of the object in the air.

### Actual results

print will fail because of missing support

### Expected results

that the same support is available no matter if "Use Adaptive Layer" is enabled or disabled.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

After checking Cura 4.13.1, I can tell thet there is the same issue.

Skybeat, 2022-05-02:
since I have seen that other bug reports have already been responded to, I wanted to ask if you have already seen my bug.

fvrmr, 2022-05-05:
Hi @Skybeat thank you for your report.
Could you share your project file (file - save project) and share it here as a .zip on GitHub.

Skybeat, 2022-05-06:
> Hi @Skybeat thank you for your report. Could you share your project file (file - save project) and share it here as a .zip on GitHub.
I tried but i get all the time the message "is not included in the list" after I add the zip to my comment in github.

Skybeat, 2022-05-06:
Attached my project as ZIP

Skybeat, 2022-05-06:
sorry I clicked the wrong button and closed the ticket. 
I can't add the zip even if I trie in differend browsers, I get all the time the message  "is not included in the list"

Skybeat, 2022-05-06:
since I'm not able to add the zip directly in GITHUB I have added it to my onedrive. You can dowloaded the zip from this link
[https://1drv.ms/u/s!AnayxfPV8dxsivkeG3DfCPWvINAH9Q?e=ap7E2C](url)

closed 2022-05-23 by github-actions[bot]

Skybeat, 2022-05-23:
@nallath thank you ;-)

nallath, 2022-06-10:
Sorry that it took so long! I've been able to reproduce your issue and I will discuss it with the rest of the team.

Skybeat, 2022-06-15:
> Sorry that it took so long! I've been able to reproduce your issue and I will discuss it with the rest of the team.

you are welcome and thank you for the update.

nallath, 2022-06-21:
We already had a ticket that mentions this issue on our backlog. It turns out that it happens for skin as well (which is the original thing it was reported on!). That also explains why I wasn't able to find it in the first place.

For future reference: CURA-5104