Ultimaker/Cura #12019: Repeatable segfault in SkeletalTrapezoidationGraph::collapseSmallEdges

kdkd, 2022-04-29:
### Application Version

5.0.0-beta+1

### Platform

MacOS 12.3.1 M1

### Printer

 Prusa I3 Mk3

### Reproduction steps

I have a project that will crash during slicing every time. It's not a public model so I can't attach the .3mf file here, but I can send it to someone who wants to reproduce it.

### Actual results

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000040
Exception Codes:       0x0000000000000001, 0x0000000000000040
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [1556]

VM Region Info: 0x40 is not in any region.  Bytes before following region: 4371582912
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      104911000-104b05000    [ 2000K] r-x/r-x SM=COW  ...OS/CuraEngine

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   CuraEngine                    	       0x1049c30f9 cura::SkeletalTrapezoidationGraph::collapseSmallEdges(long long) + 2425
1   CuraEngine                    	       0x104998eaf cura::SkeletalTrapezoidation::constructFromPolygons(cura::Polygons const&) + 2223
2   CuraEngine                    	       0x1049984a8 cura::SkeletalTrapezoidation::SkeletalTrapezoidation(cura::Polygons const&, cura::BeadingStrategy const&, cura::AngleRadians, long long, long long, long long, long long) + 200
3   CuraEngine                    	       0x104a08b16 cura::WallToolPaths::generate() + 1718
4   CuraEngine                    	       0x1049f99d5 cura::WallsComputation::generateWalls(cura::SliceLayerPart*) + 1669
5   CuraEngine                    	       0x1049fa0cb cura::WallsComputation::generateWalls(cura::SliceLayer*) + 59
6   CuraEngine                    	       0x1049421b8 cura::FffPolygonGenerator::processBasicWallsSkinInfill(cura::SliceDataStorage&, unsigned long, std::__1::vector<unsigned long, std::__1::allocator<unsigned long> > const&, cura::ProgressStageEstimator&) + 440
7   CuraEngine                    	       0x1049418c5 cura::FffPolygonGenerator::slices2polygons(cura::SliceDataStorage&, cura::TimeKeeper&) + 1573
8   CuraEngine                    	       0x10493fca3 cura::FffPolygonGenerator::generateAreas(cura::SliceDataStorage&, cura::MeshGroup*, cura::TimeKeeper&) + 51
9   CuraEngine                    	       0x104995976 cura::Scene::processMeshGroup(cura::MeshGroup&) + 806
10  CuraEngine                    	       0x1049cd8fb cura::Slice::compute() + 107
11  CuraEngine                    	       0x104a138b3 cura::ArcusCommunication::sliceNext() + 1699
12  CuraEngine                    	       0x10491570c cura::Application::run(unsigned long, char**) + 748
13  CuraEngine                    	       0x104914841 main + 81
14  dyld                          	       0x204e2651e start + 462

### Expected results

Slicing without crashing

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log.zip](https://github.com/Ultimaker/Cura/files/8591306/cura.log.zip)
[crash report.txt](https://github.com/Ultimaker/Cura/files/8591313/crash.report.txt)

fvrmr, 2022-05-02:
Hi @kdkd thank you for your report.
Could you send the file to f[dot]vermeer[at]ultimaker[dot]com? Then I can have a look and see if I can reproduce your issue.

fvrmr, 2022-05-03:
I linked your issue to an already existing ticket on our backlog.
Devs see CURA-9178

Hello1024, 2024-10-31:
Note this seems to happen more often when I have "Mesh Fixes -> Maximum deviation" set to a big number.

I suspect it might be a bug triggered by bad polygons where two lines are coincident or very nearly coincident.

A workaround when one sees this bug is to slightly adjust the maximum deviation and re-slice and it won't crash next time around.