Ultimaker/Cura #12020: Missing Python 3.10 and libm.so.6 in AppImage

hjgode, 2022-04-29:
### Application Version

5.0.0-beta+1

### Platform

Debian 10

### Printer

None

### Reproduction steps

Download released beta Ultimaker-Cura-5.0.0-beta+1-linux.AppImage onto Debian 10 (64Bit) system.

Change to executable: chmod a+x Ultimaker-Cura-5.0.0-beta+1-linux.AppImage

Try start Ultimaker-Cura-5.0.0-beta+1-linux.AppImage using "./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage"

### Actual results

Error:

└──╼ $./Ultimaker-Cura-5.0.0-beta+1-linux.AppImage 
[55755] Error loading Python lib '/tmp/.mount_UltimaEbRp5k/libpython3.10.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_UltimaEbRp5k/libpython3.10.so.1.0)

Does not start.

### Expected results

Ultimaker Cura 5.0.0+beta+1 should start without error

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

4.19.0-18-amd64

no log for this in ~/.local/share/cura...

AppImage should come with all required dependencies

jellespijker, 2022-05-01:
Cura 5.0 requires a Linux system with atleast glibc 2.31, Debian 10 is shipped with glibc 2.28. This means that Debian 11 is the first version which can support Cura 5.0

@fvrmr Is this mentioned in the changelog? See also #11863

Ghostkeeper, 2022-07-15:
We have a potential solution for this in Cura 5.1. Did you try the beta?

Or else, one of the Linux builds here: https://github.com/Ultimaker/Cura/actions/runs/2676549012

hjgode, 2022-07-16:
On Debian 11 it is working. You should add a requirements list and mention Debian 11. Please close this.

hpsaturn, 2022-07-16:
> We have a potential solution for this in Cura 5.1. Did you try the beta?
> 
> Or else, one of the Linux builds here: https://github.com/Ultimaker/Cura/actions/runs/2676549012

Thanks, but it fails with the same output:

```bash
 ./Ultimaker-Cura-5.1.0-beta+cura_8665_119-Linux-X64.AppImage 
[11875] Error loading Python lib '/tmp/.mount_Ultima8G83SR/libpython3.10.so.1.0': dlopen: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_Ultima8G83SR/libpython3.10.so.1.0)
```

hpsaturn, 2022-07-16:
> On Debian 11 it is working. You should add a requirements list and mention Debian 11. Please close this.

well, I think that the solution or answer to this can't be "Debian 11 it is working". The issue is clear: "Cura 5.0 won't run on Debian 10", that is to say, the issue can't closed. I think.

jellespijker, 2022-07-16:
Cura 5.x won't work on any Debian 10 system because it doesn't meet the minimum requirements. See my earlier answer.

> Cura 5.0 requires a Linux system with atleast glibc 2.31, Debian 10 is shipped with glibc 2.28. This means that Debian 11 is the first version which can support Cura 5.0

Since it is unlikely that we will ever have the resources to fix this I will close this ticket.

Although I (we) would have loved to keep supporting older OSes, we simply can't do that here, due to our upgrade to PyQt6, which was needed to support newer architectures, such as Apple M1.

closed 2022-07-16 by jellespijker