Ultimaker/Cura #12021: Closing machine settings with cursor still in active setting does not save that setting.

Wesztman, 2022-04-29:
### Application Version

 Ultimaker-Cura-5.0.0-beta+1-win64.exe

### Platform

Windows 10

### Printer

Ender 3

### Reproduction steps

1. Go into "Machine Settings".
2. Change setting "Y (Depth)".
3. Close Machine setting without putting cursor in any other text box.

### Actual results

The changed setting is not saved.

### Expected results

The setting should be saved.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

If moving the cursor to a new box before closing the settings window the setting is saved as it should.

fvrmr, 2022-05-03:
Hi @Wesztman thank you for your report.
I have discussed this with the team and it will not get a high priority for now, so it will not be fixed in 5.0 stable. But we have a ticket on our backlog for this. 
Devs see CURA-9240

MrBill57, 2022-05-03:
Thanks! I'm sure there is a list of things that will be fixed.
Amazing program!
I'll be patient :-)

fvrmr, 2022-05-16:
This is fixed with https://github.com/Ultimaker/Cura/pull/12116, so it is in the 5.0 stable release!

closed 2022-05-16 by fvrmr