Ultimaker/Cura #12023: Issues with plugins

leoheck, 2022-04-29:
### Application Version

5.0

### Platform

Linux

### Printer

none

### Reproduction steps

Launch cura

### Actual results

[image]

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

jedwards1230, 2022-04-30:
Are you using the latest dev version of the [OctoPrintPlugin](https://github.com/fieldOfView/Cura-OctoPrintPlugin/releases) that supports the Cura 5.0 beta?

leoheck, 2022-05-01:
Curs is asking to uninstall old plugins. This is not working.

closed 2022-05-02 by fvrmr