Ultimaker/Cura #12024: Install successful, but when Cura 5.0 is started, files are missing

KevinGroninga, 2022-04-29:
### Application Version

5.0.0

### Platform

Windows 7 Home Premium

### Printer

Ender 3 v2

### Reproduction steps

1. Downloaded and installed the Win64 version of Cura 5.0.0 beta 1
2. During install everything appears fine.
3. Start Cura 5.0.0 and I encounter 2 errors about missing files and Cura cannot start.
[image]
[image]

### Actual results

Cura 5.0.0 fails to start because of files missing.

### Expected results

Cura 5.0.0 should start and allow me to import and slice STL files.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

fieldOfView, 2022-04-30:
Unfortunately, Cura 5 will not run on Windows 7. In order to support newer (current) versions of operating systems, support for older versions (that are no longer supported by their respective vendors) was dropped. 

Cura is built using Python and Qt, and the current version of neither Python nor Qt support Windows 7.

fvrmr, 2022-05-02:
Hi @KevinGroninga thank you for your report.
Like @fieldOfView mentioned we don't support Windows 7 anymore. The latest Cura version that is supported on Windows 7 is Cura 4.10, which you can find here: https://github.com/Ultimaker/Cura/releases/tag/4.10.0

I will close this issue since this is not something we will change.

closed 2022-05-02 by fvrmr