Ultimaker/Cura #12025: Octoprint Plugin

stonecold31666, 2022-04-30:
### Application Version

5.0

### Platform

Windows 11

### Printer

Ender 3 V2

### Reproduction steps

Uninstalled and reinstalled after restart and still have no Octoprint options. 

### Actual results

i have to save to disk then uploaded to octoprint manually

### Expected results

after slicing optioon to print to octoprint. worked fine in 4.x. no plugins in the market place

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8595425/cura.log)
[image]
[image]
[image]

jedwards1230, 2022-04-30:
> If you are missing your favorite plug-in in the Cura 5.0 Beta, please allow the developers some time to update their plugin to be compatible with the Beta. For older versions of Cura you can download your favorite plugins from Cura itself by selecting 'Marketplace' in the upper right corner.

It will take time for developers to update their plugins for this release of Cura.

The [OctoPrintPlugin](https://github.com/fieldOfView/Cura-OctoPrintPlugin) repo has a development version of the plugin that shows support for this beta version of Cura.

fieldOfView, 2022-04-30:
Prereleases for most of my plugins (including the OctoPrint Connection plugin) can be found here:
https://community.ultimaker.com/topic/40313-fieldofview-plugins-for-cura-50/

closed 2022-05-02 by nallath