Ultimaker/Cura #12026: Random under extrusion near z-seam

LasagnaSam, 2022-04-30:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 3 Pro

### Reproduction steps

I tried printing a few things in Cura 5.0 beta out of curiosity. I am printing with petg at 240 degrees Celsius on an ender 3 pro. I used my normal Cura profile for petg from cura 4.13.1 which usually gives me good prints. 

### Actual results

When I print something in Cura 5.0 beta I get a rough surface near the z-seam. It seems that there is under extrusion near this region.

### Expected results

Extrusion should be consistent and surface should be fairly smooth even near the seam. When I slice the exact same model with the exact same slicer profile in Cura 4.13.1 I get a normal looking surface.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
21a721e5ad3e.jpg)
[image]
[image]

fvrmr, 2022-05-02:
Hi @LasagnaSam thank you for your report.
Could you share your project file (file - save project) and share it here as a .zip on GitHub.

LasagnaSam, 2022-05-02:
I just added the project file for the model I sliced in Cura 5.0

closed 2022-05-23 by github-actions[bot]

skl111, 2022-05-24:
Any feedback, I am experiencing this as well. various brands of petg around 240/245. I see a model was uploaded, did it show anything?

nallath, 2022-05-24:
Bad bot!

nallath, 2022-05-24:
Sorry that this got closed! The bot was broken for a while and it never removed the label. I fixed the bot yesterday, so it closed it

Arne-W, 2022-06-15:
I have the same issue on an Anycubic Vyper. Since I changed to 5.0.0 I have problems with under extrusion near z-seam. I did a lot of experimenting but was not able to fix it.
For now I will switch back to 4.13 an check if that fixes the issue fro me.

skl111, 2022-06-15:
If you try wall order setting and change it from "outside to inside" to
"inside to outside" that should make it work like 4.x