Ultimaker/Cura #12032: Initial Layer: Brim only

DoganM95, 2022-04-30:
### Is your feature request related to a problem?

First layers can have tiny circles/lines etc. These are very hard to stick to the bad becuase of their size and most of the times get loose and ruin the print.

### Describe the solution you'd like

Cura should offer filling the complete initial layer with brim only, meaning no tiny inner objects like lines, circles, etc. instead it should create the complete first layer from the most outer wall to the center in one go, without stopping, without any gaps in between. This would create a 100% infill first layer to make the first layer stick perfectly and giving the chance to add tiny parts to the bottom of the object.

### Describe alternatives you've considered

- Gluestick for first layer (mess)
- Increasing temps (mess)

### Affected users and/or printers

everyone, who has small parts in their print will benefit. Afrer print, it will only require a quick post-processsing (by hand), cutting the excessive brim off the first layer.

### Additional information & file uploads

This is the current situation, brim (turquoise) is only used on the outside. Small rings (green) will easily lift and mess up the whole print.
[image]

This is at least in color what i mean. The whole layer should be printed in one go, in this case in circular motions from the outside to the inside. It should only take the most outer wall of the part to print as a reference, crete an outer brim and fill the inside of the first layer completely.
[image]

nallath, 2022-06-15:
So like a raft?

DoganM95, 2022-06-17:
Yes, seems like the Raft was exactly what i was looking for. Thanks alot!

closed 2022-06-17 by DoganM95