Ultimaker/Cura #12033: Multiply parts not working without changing count

hawkeyexp, 2022-04-30:
### Application Version

5.0 beta 1

### Platform

Windows 11

### Printer

Voron 2.4

### Reproduction steps

- Place part on print bed
- Right click on part
- Select multiply parts
- Leave count as is (1)
- Click ok

### Actual results

After clicking ok no additional part is added.

Solution: do same steps and click count + to get two and click count - to go back to one - now a part is added after clicking ok - seems a bug with pre selected value.

### Expected results

A second part should be appear

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

not additional files needed to reproduce - only a clean start and the effect happens

hawkeyexp, 2022-04-30:
Seems to be possible solved with commit d12703289b8c169e50421b4c0d548b3611422889 from yesterday :-)

fieldOfView, 2022-04-30:
Indeed. You can close this duplicate issue.

closed 2022-04-30 by hawkeyexp