Ultimaker/Cura #12036: Interface hangs instantly when viewing opening custom slicing or other settings

halr9000, 2022-04-30:
### Application Version

5.0-beta1

### Platform

win 11

### Printer

Snapmaker 2 A350, Creality Ender 3 v2

### Reproduction steps

Preface
1. Uninstall 4.latest (note: have had very similar issues OFTEN in past, making the app all but unusable, but I keep coming back to try and get this resolved)
2. Install 5.0-beta-1
3. Load any model
4. Sliced successfully

Case 1

1.  Attempt to change support type or any other setting outside of the default profile -> click custom, then start typing "tree" in the search box -- immediate hang

Case 2
1. Restart, go to custom, then open profile filter settings. 
2. Check any advanced setting
3. Click ok, immediate hang

Case 3
1. Change profile filter from basic to advanced, hang

Case 4
1. Go to settings menu, it will oepn
2. Click to switch to any other tab in the settings dialog, immediate hang

### Actual results

In every case, the UI will hang for many seconds, but eventually control will return. Unfortunately, it's super easy to trigger the behavior again while trying to get out of the dialog I was hung in, so it's very frustrating and thoroughly unusable.

If I try to do anything in the UI, Windows will see the thread not responding and prompt to close program as shown below. 

### Expected results

No hangs, ever.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[image]

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/8597509/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/8597510/stderr.log)

halr9000, 2022-04-30:
Also note minor UI bugs which I don't think are related as seen in the above 2nd image. Text overflow and buttons borders look wrong.

nallath, 2022-06-15:
This has been fixed for 5.0.0 final

closed 2022-06-15 by nallath