Ultimaker/Cura #12037: Improper # of walls in Beta 5.0

tasmaine, 2022-04-30:
### Application Version

5.0.0 (Beta)

### Platform

Windows 10

### Printer

Own printer profile (modified Creality Cr6-SE)

### Reproduction steps

Attempting to print a cylinder with a 2.4 mm wall thickness.  Outside diameter approx 210 mm.  Nozzle selected was 0.8mm.  Line width 0.8 mm. The intent is to have three wall thicknesses.  Also replicated results using the same file with 2.1 mm wall thickness, 0.8mm nozzle, and 0.7mm line width.

### Actual results

Only 2 wall thicknesses were printed in each case.  This is even when I explicitly input 3 wall thicknesses

[image]

### Expected results

Three walls should have been printed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

N/A

fvrmr, 2022-05-02:
Hi @tasmaine thank you for your report.
Could you share your project file (file - save project) and share it here in a .zip on GitHub. 
With a project file it is easier for me to reproduce your issue and see what is happening.

tasmaine, 2022-05-02:
Sorry for the confusion regarding closing this issue.  It was an inadvertent click.  Here is the project file
[CCR6SE_TrueCircle_Wastebasket.zip](https://github.com/Ultimaker/Cura/files/8602932/CCR6SE_TrueCircle_Wastebasket.zip)
.

tasmaine, 2022-05-23:
How was this closed.  I provided the requested info 3 weeks ago?

tasmaine, 2022-05-23:
You need better bots, if they close issues when the requested info has been provided.  Please see this link from 3 weeks ago.
https://github.com/Ultimaker/Cura/issues/12037#issuecomment-1114815431

nallath, 2022-06-10:
The bot stopped working for a while. As such it didn't remove the label when you replied. We kicked it back into life, so that's why it was suddenly behaving a bit weird.

nallath, 2022-06-10:
Well, the model you posted doesn't have a wall thickness of 2.4, it has a wall tickness of 2mm.

Also note that with Arachne, it can change the wall thickness (and the number of walls) in certain cases.

closed 2022-06-10 by nallath