Ultimaker/Cura #12039: Linux Beta 5.0 Missing options for Creality E3 printer

geekthesteve, 2022-04-30:
### Application Version

5.0.0 Beta+1

### Platform

Ultimaker-Cura-5.0.0-beta+1-linux.AppImage

### Printer

Creality E3

### Reproduction steps

1. Try to slice an STL file created from FreeCAD 
2. Read error message

### Actual results

Octoprint reported Some slicer settings missing from your gcode file.  Missing Settings: Retraction Length, Retraction Speed, Deretraction Speed, Z Lift Height, X/Y Travel Speed, Travel Speed.  

Windows version of same Beta works OK.  

### Expected results

Should:  Linux version should work OK, like  the Windows version does.  

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2022-05-02:
Hi @geekthesteve thank you for your report.
Are you sending your gcode file via Octoprint? Could you share your gcode file?

fieldOfView, 2022-05-02:
This is not an issue with Cura, but you did not follow the Octolapse configuration instructions.
See https://github.com/FormerLurker/Octolapse/wiki/V0.4---Automatic-Slicer-Configuration

geekthesteve, 2022-05-02:
You are correct, I did not follow the Octolapse configuration instructions.  This was not required on the Windows V5.0 Beta but that could be because I did those steps on Windows long, long ago and they were carried forward from previous Cura versions.  

This Linux version of Cura V5.0 Beta is on a fairly fresh Linux install and I could find no way of carrying over the Windows configuration to Linux.  I guess not many people have this need since I could not find a script or guide on how to replicate from Windows to Linux only a note that there are minor differences in file format between the two platforms that make this impossible.  

Thanks ever so much for the link, I will perform the updates required and look forward to going from design to printing on one platform just like I have on Windows.  BTW I very much like the 5.0 Cura and see marked differences in printing quality on the same part between the two versions.  

I assume this issue will be closed from here on.  

Best,
Steve

geekthesteve, 2022-05-02:
User error, see comments.

closed 2022-05-02 by geekthesteve