Ultimaker/Cura #12041: Multiply only works when set to 2 or more

EA2BA, 2022-04-30:
### Application Version

5.0.0

### Platform

Windows 11

### Printer

Ender 3 Pro

### Reproduction steps

Import a STL to slice
See the import object
Right click and select multiply
Leave it set as 1 hit done
You get nothing
Repeat but select 2 (you get 2 objects)

I found you can pick an object and multiply by 1 (create 1 copy)
It only works with 2 minimum then you have to delete the extra 1 to have just a single copy

### Actual results

You can’t select 1 and get a single copy

### Expected results

You can leave it set to 1 and get a single copy

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Happens and ALL imported object/s wether new or old

fieldOfView, 2022-04-30:
Thanks for reporting. This has already been fixed for the full release.

GregValiant, 2022-05-01:
And in the interim if you change it to 2 and then back to 1 it should be OK.  There are other dialogs with this same sort of problem where you have to click somewhere else to set the change (StartUp Gcode box in Machine Settings is one).

fieldOfView, 2022-05-01:
> There are other dialogs with this same sort of problem where you have to click somewhere else to set the change (StartUp Gcode box in Machine Settings is one).

I hope you reported those? I fixed only the Multiply dialog.

GregValiant, 2022-05-01:
I think the Machine Settings dialog was covered in #12021.  I seem to recall another one regarding missing buttons in some dialogs.  Machine Settings was one of them affected.  The only way to dismiss the dialog is the "X" on the titlebar.  I don't know if that would be a bug or if it's just non-intuitive.

closed 2022-05-02 by fvrmr