Ultimaker/Cura #12042: "Plugin" could not be removed, had to edit packages.json directly

OystanX1, 2022-05-01:
### Application Version

5.0.0-beta+1-linux

### Platform

Ubuntu 20.04 / Linux

### Printer

Not relevant

### Reproduction steps

Removed printer Dremel 3D45 and plugin for Dremel printers.

Then after restart of Cura, got a message that said pluging not properly removed.

Traced the issue to
$Home/.local/share/Cura/5.0/packages.json 

Edited it to see below and saved

{
    "dismissed": [],
    "installed": {},
    "to_install": {},
    "to_remove": [],
    "version": 1
}

Restart Cura problem solved.

IMO: The uninstall of the plugin should do this automaticly ... 
Not a serious bug, but a bit annoying ...

### Actual results

Plugin not removed properly message

### Expected results

No annoying messages

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8598339/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/8598340/stderr.log)

fvrmr, 2022-05-02:
Hi @OystanX1 thank you for your report.
Did you get a message with uninstalling errors? See https://github.com/Ultimaker/Cura/issues/11865 or did you get an other message?

OystanX1, 2022-05-04:
Hi,

I got a message telling me that the plugging wasn't removed or 
something.

The reason was that inside the file "packages.json" there was a 
redundant text / not removed by plugin uninstall.

I fixed the problem by editing the packages.json file with a text editor 
(just as an attempt, did not know if it would work, but it did ...)

Off topic

My removal of the Dremel plugin is permanent. Dremel has given up the 3D 
printer market and can't/refuse to supply parts.

My Dremel 3D45 printer got problems and seller unable to repair caused 
by Dremels actions. Got my money back and went for

a Flashforge Guider 2s printer with direct extruder. I wish Ultimaker 
was an option too but the Bowden tube technology is

incompatible with e.g. LW-PLA / LW-ABS (foaming extruder materials) as 
far asI know.

Br,

Odd H.

Den 2022-05-02 10:51 skreiv Fenne:

> Hi @OystanX1 [1] thank you for your report.
> Did you get a message with uninstalling errors? See #11865 [2] or did 
> you get an other message?
> 
> [4].

Links:
------
[1] https://github.com/OystanX1
[2] https://github.com/Ultimaker/Cura/issues/11865
[3] 
https://github.com/Ultimaker/Cura/issues/12042#issuecomment-1114633102
[4] 
https://github.com/

closed 2022-05-23 by github-actions[bot]