Ultimaker/Cura #12052: Cura 5.0.0 beta hangs when slicing

CakePiper69, 2022-05-02:
### Application Version

5.0.0 beta

### Platform

Windows 10

### Printer

Monoprice Select Mini V2

### Reproduction steps

Opened a jpg file to print as a lithophane.
Then opened an stl file to print as a lithophane.
In both cases, at the loaded size (100%) the slicing would start, but hang at about 25%.
If I reduced the size of the model to 50%, then it would slice.  If I then increased the model back to 100%, it would slice.

I tried loading other stl files (like Benchy, the tugboat) and they would slice just fine.

### Actual results

sometimes the file could be sliced and sometimes it would not slice.  On several occasions, the little blue bar would go back and forth for over ten minutes with no progress.  Other times, it would slowly progress and complete the slicing.

### Expected results

When the jpg or stl file is opened, it should slice.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/8600629/cura.log)
[MSMV2E3D_Matt and Kate 5x7 edited framed.zip](https://github.com/Ultimaker/Cura/files/8600634/MSMV2E3D_Matt.and.Kate.5x7.edited.framed.zip)
[image]

fvrmr, 2022-05-02:
Hi @CakePiper69 thank you for your report.
I can reproduce your issue. I will bring it up with the team. Keep you posted.

Ghostkeeper, 2022-05-04:
Love the photo!

This is a traceback that I got from the error, and some debugging commands at the bottom:

```
Thread 13 "CuraEngine" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff1a08640 (LWP 12239)]
0x00005555557a72ca in cura::SkeletalTrapezoidation::separatePointyQuadEndNodes (this=0x7ffff1a07940) at /home/trin/Gedeeld/Projects/CuraEngine/src/SkeletalTrapezoidation.cpp:472
472	            quad_start->twin->to = new_node;
(gdb) bt
#0  0x00005555557a72ca in cura::SkeletalTrapezoidation::separatePointyQuadEndNodes (this=0x7ffff1a07940) at /home/trin/Gedeeld/Projects/CuraEngine/src/SkeletalTrapezoidation.cpp:472
#1  0x00005555557a7024 in cura::SkeletalTrapezoidation::constructFromPolygons (this=0x7ffff1a07940, polys=...) at /home/trin/Gedeeld/Projects/CuraEngine/src/SkeletalTrapezoidation.cpp:437
#2  0x00005555557a68e6 in cura::SkeletalTrapezoidation::SkeletalTrapezoidation (this=0x7ffff1a07940, polys=..., beading_strategy=..., transitioning_angle=..., discretization_step_size=800, transition_filter_dist=100000, allowed_filter_deviation=100, 
    beading_propagation_transition_dist=350) at /home/trin/Gedeeld/Projects/CuraEngine/src/SkeletalTrapezoidation.cpp:361
#3  0x0000555555718d53 in cura::WallToolPaths::generate (this=0x7ffff1a07ae0) at /home/trin/Gedeeld/Projects/CuraEngine/src/WallToolPaths.cpp:107
#4  0x0000555555719925 in cura::WallToolPaths::getToolPaths (this=0x7ffff1a07ae0) at /home/trin/Gedeeld/Projects/CuraEngine/src/WallToolPaths.cpp:200
#5  0x000055555583943b in cura::WallsComputation::generateWalls (this=0x7ffff1a07c80, part=0x7fffe02f5bf0) at /home/trin/Gedeeld/Projects/CuraEngine/src/WallsComputation.cpp:73
#6  0x000055555583989f in cura::WallsComputation::generateWalls (this=0x7ffff1a07c80, layer=0x555556464988) at /home/trin/Gedeeld/Projects/CuraEngine/src/WallsComputation.cpp:89
#7  0x000055555578aecb in cura::FffPolygonGenerator::processWalls (this=0x5555559e5140 <cura::FffProcessor::instance+4064>, mesh=..., layer_nr=43) at /home/trin/Gedeeld/Projects/CuraEngine/src/FffPolygonGenerator.cpp:730
#8  0x000055555578f8d1 in _ZN4cura19FffPolygonGenerator27processBasicWallsSkinInfillERNS_16SliceDataStorageEmRKSt6vectorImSaImEERNS_22ProgressStageEstimatorE._omp_fn.0(void) () at /home/trin/Gedeeld/Projects/CuraEngine/src/FffPolygonGenerator.cpp:470
#9  0x00007ffff7f57b9e in ?? () from /lib/x86_64-linux-gnu/libgomp.so.1
#10 0x00007ffff72afb43 in start_thread (arg=<optimized out>) at ./nptl/pthread_create.c:442
#11 0x00007ffff7341a00 in clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:81
(gdb) print quad_start
$1 = (cura::SkeletalTrapezoidation::edge_t *) 0x7fffbce98b20
(gdb) print quad_start->twin
$2 = (cura::HalfEdge<cura::SkeletalTrapezoidationJoint, cura::SkeletalTrapezoidationEdge, cura::STHalfEdgeNode, cura::STHalfEdge>::edge_t *) 0x0
```

Technical babble: To create the walls, CuraEngine is now creating a skeletal graph of the polygon of a layer. This skeletal graph is supposed to be a doubly-connected edge list, where each segment consists of two edges, one going back and one going forth. These two edges connect up with their neighbouring edges to allow traversing the graph.
It appears that during the creation of the walls it encountered a degeneracy. As a result it didn't create a twin edge for one of the edges in the skeletal graph. Then later in the process it's trying to navigate through that graph and crashes when it tries to access that twin.
We've seen this issue before in a couple of prints. We'll be adding it to the test cases of the bug. It's been considered a critical issue and we're trying to fix it before the 5.0 stable release. It's a pretty hairy problem though.

Developers, refer to issue CURA-9178.