Ultimaker/Cura #12057: Infill, Top & Bottom not filling

pierremilcent, 2022-05-02:
### Application Version

5.0 Beta

### Platform

MacOS M1

### Printer

Artillery Sidewinder X2

### Reproduction steps

i am not able to get the same result with Cura 5 and Cura 4.13.1 for the same part. Cura 5 does not allow me to have infill done, it keeps printing 2 walls and nothing inside.

What should i check in Cura 5 parameters that influence this behaviour ? 

Thanks
[image]
[image]
.

### Actual results

[image]

### Expected results

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

fvrmr, 2022-05-02:
Hi @pierremilcent thank you for your report.
Could you check if the infill is enabled in preview mode?
[image]

If that is not the case, please share your project file (file - save project) as a .zip here on Github.

pierremilcent, 2022-05-02:
Yes it is checked. 2 screenshots to illustrate. I also add 3mf files for both cura versions.
[image]
[image]
[3mf.zip](https://github.com/Ultimaker/Cura/files/8604861/3mf.zip)