Ultimaker/Cura #12061: Fala ao fatiar

oc-leite, 2022-05-02:
### Application Version

ultimaker cura 5 beta

### Platform

Windows 10

### Printer

Creality ender3

### Reproduction steps

1. Importei e  dimensionei o objeto na mesa de impressão
2. Escolhi as configurações

### Actual results

erro de fatiamento após alguns minutos

### Expected results

o resultado esperado seria o fatiamento do modelo

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
[cura.log](https://github.com/Ultimaker/Cura/files/8602590/cura.log)

fvrmr, 2022-05-02:
Hi @oc-leite thank you for your report.
Could you share your project file (file - save project) here as a .zip on Github?

oc-leite, 2022-05-02:
[base under hulk.zip](https://github.com/Ultimaker/Cura/files/8603059/base.under.hulk.zip)

fvrmr, 2022-05-04:
Thank you, unfortunately I can't reproduce your issue with an stl file. Could you share a project file (file - save project)?

closed 2022-05-23 by github-actions[bot]