Ultimaker/Cura #12063: Issue with tree supports in Cura V5 Beta

Nabla78, 2022-05-02:
### Application Version

5.0.0-beta+1

### Platform

Windows 10

### Printer

Artillery X1

### Reproduction steps

Just slicing.
There is some little issue while generating tree supports (support not needed on backside).

[Cura V5 bug.zip](https://github.com/Ultimaker/Cura/files/8604917/Cura.V5.bug.zip)

### Actual results

Tree support generation without any reason (nothing to support).

### Expected results

Just usefull tree support

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Everything in Zip file

fvrmr, 2022-05-04:
Hi @Nabla78 thank you for your report.
We have a ticket on our backlog related to your issue. I will add your project file as a test case to the ticket.
For internal reference see CURA-9170

DerGenaue, 2023-07-04:
FYI: This probably has been solved in the new Cura 5.4 release