Ultimaker/Cura #12066: Failed Slice with raft and supports with minimal part area touching platform

truegret, 2022-05-02:
### Application Version

5.0.0-beta+1

### Platform

Windows 11

### Printer

Creality-CR-10S

### Reproduction steps

1. Imported complex Alastor__-_Head.stl from Thingiverse: 5322398
2. Created printer profile with 0.3mm nozzle
3. With ALL settings enabled/available, modified **Printer Settings|Nozzle Diameter** from **0.4 to 0.3**
4. Modified **Quality|Line Width** from **0.48 to 0.36**
5. Positioned model such that minimal features are touching buildplate
6. Modified **Build Plate Adhesion|Build Plate Adhesion Type** from **Brim to Raft**
7. Clicked Slice

### Actual results

At roughly 50-60% into slicing, received a popup:
Slicing failed
Slicing failed with an unexpected error. Please consider reporting a bug on our issue tracker.

### Expected results

Expected sliced model with supports on a raft.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Attached (uploaded) cura.log, screenshots, and project file of this issue.

fvrmr, 2022-05-04:
Hi @truegret thank you for your report.
I can reproduce your issue, will bring it up with the team. Keep you posted.

fvrmr, 2022-05-05:
I created a ticket on our backlog for this. 
Devs see CURA-9251

truegret, 2022-05-07:
I attempted a calibration test with a Benchy STL, and smaller nozzle ( 0.2mm ).  I earned the "Slicing failed" error with this, too.  I don't see a means of uploading those related files without creating another bug.

Reproduction steps
Imported complex Alastor__-_Head.stl from Thingiverse: 763622
Created printer profile with 0.2mm nozzle
With ALL settings enabled/available, modified Printer Settings|Nozzle Diameter from 0.4 to 0.2
Modified Quality|Line Width from 0.48 to 0.36
Modified Quality|Layer Height from 0.16 to 0.12
Modified Quality|Initial Layer Height from 0.20 to 0.16
Modified QUality"Initial Layer Line Width from 140.0% to 100.0%
Positioned model such that minimal features are touching buildplate
No Raft
Brim
Clicked Slice

Ghostkeeper, 2022-06-16:
This was analysed, and we found it to be a duplicate of https://github.com/Ultimaker/Cura/issues/12312. As a workaround, you can disable the Randomize Infill Start setting.