Ultimaker/Cura #12073: Print Setting NOT being over ridden by Material Change

HappyMonkey666, 2022-05-03:
### Application Version

5.0.0

### Platform

Windows 10

### Printer

Creality Ender 6

### Reproduction steps

Uninstall all instances of CURA, Delete App Date Local and Roaming and Cache.
New Install of Cura 5.0.0

Open Cura, Load test file 
Material Selected Generic PLA 
Print Setting ANY [0.12 0.16 0.2 0.28]
Slice
Save to Disk

### Actual results

Nozzle and Bed temperatures are FIXED and will not change to Material Settings

Nozzle Temperature NOT Correct its 220C should be 200C
M104 S220
Bed Temperature NOT correct its 50C should be 60C
M140 S50

### Expected results

The correct Nozzle and Bed temperatures should be set according to the Material Selected

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8612367/cura.log)

Asterchades, 2022-05-04:
Any settings which have been manually changed in the print properties will take precedence over the material properties. It's always been this way. The reason that the material properties work is because placeholder tokens get used (eg {material_print_temperature}), but you can't manually enter those and printer profiles can also override that default.

In this specific case, the printer profile is indeed overriding your set value. I couldn't even begin to guess _why_ it's being told to, but whoever submitted the Ender-6 profile decided that people would want to print at 220°c on the nozzle and 50°c on the bed - regardless of what material is in use. And it's been this way since before 5.0 was released. So you would need to adjust the temperatures in the print properties to use what you're after, rather than in the material properties.

Specifically: "creality_ender6.def.json"
line 17: "material_print_temperature": { "value": 220 },
line 18: "material_bed_temperature": { "value": 50 },

_Note that Ultimaker - the Cura team - doesn't maintain 3rd party profiles. They're maintained by community submissions._

Wesztman, 2022-06-10:
@Asterchades 

Are there any variable reference documentation as to what different variables which can be used? 

Do these variables have to be input in the json and not the gui?

Asterchades, 2022-06-10:
The only places which accept placeholder values are the printer definitions, the start/end Gcode, and post process scripts (the source script, not the values they then take). Note that while both material properties and print settings are also stored as JSON files they don't accept the placeholders.

While I don't know of any complete list of the placeholder names which is maintained, you can see all of them in the "[fdmprinter.def.json](https://github.com/Ultimaker/Cura/blob/main/resources/definitions/fdmprinter.def.json)" file. There will also be a copy of this file in your local Cura install which will only include values appropriate for your version (<cura path>\resources\definitions for prior to v5, <cura path>\share\cura\resources\definitions for v5 onward).

JesperKrogPoulsen, 2023-03-24:
I have a question regarding this. If I choose a meterial profile, fx. PLA, the print settings are setting the nozzle temperature to what the profile says. If I change the nozzle temperature, I get the "fx" icon next to the setting to revert back to what the material profile says. If I change to my PETG material profile, the nozzle temp stays at what I changed it to, unless I click the "fx" icon and it reverts to what the PETG profile says nozzle temperature says.

The bed temperature, on the other hand, stays the same no matter what the material profile says and I don't get the "fx" icon. Why is that? What can I do to fix this?