Ultimaker/Cura #12074: Extruding on Pause at Height when using retraction

DigitalSketch, 2022-05-03:
### Application Version

5.0.0-beta+1

### Platform

Win 10

### Printer

Ender 3 V2

### Reproduction steps

Using the "Pause at height" Post Processing plugin, I set the Retraction to 500 to pull the filament out of my extruder. When I added the new filament and resumed the print, the extruder ran for a long time and pushed out a lot of filament right over the print. The nozzle moved to the position to start printing again and pushed out the extra filament.  

This happened on a production print, so I created a print test. The extrusion happened as described with the test as well. When I removed the Retraction number on the second test, no extrusion occurred. 

Attached are the gcode files used and a screenshot of the Pause at height settings for the "CE3_Me_Test.gcode" file.
 

### Actual results

After changing the filament, the nozzle moves to the position to start printing again and then extrudes a lot of filament. I'm guessing it's the same amount of filament that was retracted. 

### Expected results

After the filament is replaced and the print is resumed, I expected it to simply start printing where it left off. Or to extrude the filament at the Park Print Head X and Y,  not over the position where it should resume printing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Screenshot of the Pause at height settings used in the CE3_Me_Test.gcode file:
[image]

gcode test files I created. CE3_Me_Test.gcode file has the 500 retraction setting and the CE3_Me_Test_NoRetract.gcode has the 0 retraction setting and had no extrusion. 
[Pause_at_height_Print_Tests.zip](https://github.com/Ultimaker/Cura/files/8612721/Pause_at_height_Print_Tests.zip)

fvrmr, 2022-05-05:
Hi @DigitalSketch thank you for your report.
Could you also share your project file (file - save project) and share it here as a .zip on GitHub.

DigitalSketch, 2022-05-05:
[CE3_Me_Test.zip](https://github.com/Ultimaker/Cura/files/8631656/CE3_Me_Test.zip)

closed 2022-05-23 by github-actions[bot]

nallath, 2022-05-23:
Bad bot! This shouldn't be closed...