Ultimaker/Cura #12075: Vase mode does not respect, wall count etc.

PhilBaz, 2022-05-03:
### Application Version

5

### Platform

win 10

### Printer

Voron 2.4

### Reproduction steps

 Vase mode does not respect, wall count, bottom pattern, extra skin wall for initial layer, for initial skin layer. 

### Actual results

concentric

### Expected results

respect wall , etc. for initial skin layer.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

nothing

fvrmr, 2022-05-04:
Hi @PhilBaz thank you for your report.
This is expected behaviour with using Spiralize Outer Contour. This feature will print a solid bottom and only a single wall. 
Read here more about Spiralize Outer Contour: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/blackmagic/magic_spiralize.md

Let me know if this makes sense to you!

PhilBaz, 2022-05-04:
Im only talking about the **initial layer**. The bottom skin. The setting for, wall count, bottom pattern, extra skin wall **for initial layer**, should all have an impact. They did previously.

At the very least choice of skin pattern.

PhilBaz, 2022-05-24:
@nallath  does this need something? does the explanation of the problem make sense?

PhilBaz, 2022-05-26:
@fvrmr @nallath encountered the issue again in the new 5.0

I've added both an image and project file....

[image]
[bottom layer pattern.zip](https://github.com/Ultimaker/Cura/files/8780502/bottom.layer.pattern.zip)

PhilBaz, 2022-06-09:
There is now a duplicate report.

https://github.com/Ultimaker/Cura/issues/12391

gitlui, 2022-06-09:
As the duplicate got closed I'm commenting here to get notifications. I print a lot of boxes in vase mode and the bottom is unusable for me like this. Still using 4.13 for the time being. Hope you can fix this soon :-)

Vandresc, 2022-06-10:
Devs see CURA-7296 for our internal reference.

hapklaar, 2022-09-01:
Hi, Is this still on the radar to get fixed?

Vandresc, 2022-09-23:
This is now fixed in 5.2.

closed 2022-09-23 by Vandresc

PhilBaz, 2022-10-01:
@Vandresc 

I found new errors related to this in the 5.2 beta. You want them posted here?

PhilBaz, 2022-10-01:
@Vandresc @nallath 

here...

https://github.com/Ultimaker/Cura/issues/13420