Ultimaker/Cura #12077: Bottom Progress Bar in Preview screen

ItsJohnKelly, 2022-05-03:
### Application Version

5.0.0 Beta+1

### Platform

Win10

### Printer

Ender 5

### Reproduction steps

I loaded a model I had previously sliced in 4.13.1 with success.
I sliced it in 5 Beta
I went to preview screen and saw the vertical progress bar (Layers) on the right but there was none on the bottom edge or, the start icon. So could not run the preview in the usual way
[image]

### Actual results

cant watch preview in usual manner

### Expected results

Should have seen bottom progress bar

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/8613957/cura.log)

[5.0_20220503_193657.zip](https://github.com/Ultimaker/Cura/files/8613982/5.0_20220503_193657.zip)

fvrmr, 2022-05-04:
Hi @ItsJohnKelly thank you for your report.
Is it possible that you have enabled Force layer view compatibility mode in preferences?

closed 2022-05-23 by github-actions[bot]

ItsJohnKelly, 2022-10-11:
Oh good grief,  I have never knowingly used that option but you are correct, it WAS turned on.  Strange though that such an option should hide the horizontal progress bar.

Thanks for your help, John