Ultimaker/Cura #12078: 5.0beta - "Line1" and "line2" in the gcode

GregValiant, 2022-05-03:
### Application Version

5.0 beta

### Platform

Windows 10 Home (onboard video OpenGL 4.0 in compatibility mode)

### Printer

Creality E3Pro

### Reproduction steps

Slice a model and generate a gcode.

Within the gcode file I've started getting this (this is copied and pasted in):

;End of StartUp
;
G92 E0
**Line1
line2**
G92 E0
G92 E-2.5 ;(from search and replace)
;LAYER_COUNT:60
;LAYER:0
The printer skips by Line1 and line2 with the "unknown command" response so there is no repercussion but are those lines supposed to be something else that didn't make it in?

### Actual results

See above.

### Expected results

See above

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Line1 line2.zip](https://github.com/Ultimaker/Cura/files/8614154/Line1.line2.zip)

GregValiant, 2022-05-09:
Another example:

;End of StartUp
;
M82 ;absolute extrusion mode
Line1
line2
M83 ;relative extrusion mode
M83 ;relative extrusion mode

fieldOfView, 2022-05-09:
(whisper) 
you may want to check your extruder start and end gcode snippets
(/whisper)

GregValiant, 2022-05-09:
Geez....Good call.  They were in _both_ the Start extruder and End extruder boxes.  I was probably fooling around one night and checking something, added those, and I failed to clean that up when I was finished.

closed 2022-05-09 by GregValiant