Ultimaker/Cura #12082: [5.0 beta] Red "wall transistion length" does not disable slicing.

printingotb, 2022-05-04:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

Accidentally entered 5mm line width instead of 0.5, see project.

### Actual results

Wall transition length turns red but slicing is still allowed.

### Expected results

Slicing should be disabled or setting should not turn red.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[Uploading OpticoolerFunels.3mf.zip…]()

fvrmr, 2022-05-04:
Hi @printingotb thank you for your report.
Your project file is not correctly uploaded. Could you upload it again?

printingotb, 2022-05-04:
Try now.
[OpticoolerFunels.3mf.zip](https://github.com/Ultimaker/Cura/files/8621051/OpticoolerFunels.3mf.zip)

fieldOfView, 2022-05-04:
There's a difference between orange and red. Red means: error, can't slice with this value. Orange means: are you sure? I don't think that's a good idea, but if you say so I'll slice with this value.

fieldOfView, 2022-05-04:
To elaborate: red is normally limited to values that are physically impossible, such as a speed that is faster than the speed of light, or a layer height of 0. Orange means the value seems to exceed a normal, usable range.

nallath, 2022-05-05:
I think @printingotb is correct. Any setting in error state currently does not prevent slicing (and it should!)

fvrmr, 2022-05-05:
We have a ticket on our backlog for this!
Devs see CURA-9249

commit 509b7d65 (2022-05-05): Fix error checking never setting state to "hasError"

CURA-9249

Fixes #12082
closed 2022-05-05 by nallath