Ultimaker/Cura #12084: rotation rings not working in cura 5, unable to rotate.

davetech, 2022-05-04:
### Application Version

5.0.0

### Platform

Windows 10, ver 21H2, intel core i5 @ 3.00 ghz

### Printer

Folgertech FT5

### Reproduction steps

add an stl file to Cura 5, select the model, click on the rotate icon, no rings appear so no rotation possible.

### Actual results

Next switch off "auto fix model to floor", still no rings appear but the model can now be rotated in all three dimensions.

### Expected results

the model would no longer be fixed to the floor

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[image]
this screen shot shows my Cura 5 on the left screen and the Cura 5 service engineer's sample on the right screen.
[cura.log](https://github.com/Ultimaker/Cura/files/8619739/cura.log)

[CFFFP_3DBenchy.zip](https://github.com/Ultimaker/Cura/files/8619783/CFFFP_3DBenchy.zip)

fieldOfView, 2022-05-05:
Sounds like a duplicate of https://github.com/Ultimaker/Cura/issues/11882, which was already fixed for the final release of Cura 5.0

fvrmr, 2022-05-05:
Ah yes that is indeed related, thanks @fieldOfView!

Closing this issue since it is a duplicate!

davetech, 2022-05-05:
Hi I read what you are saying but as a standard user how do I fix this problem?

davetech, 2022-05-05:
>     Hi @davetech https://github.com/davetech thank you for your report.
>     I also can't reproduce this issue. Did you try to do a clean install of Cura? And did this happen in other Cura versions as well?

Hi Fenne,

Yes I have done a clean install but no help. This does not happen in any other version. Is there a step by step way to fix this problem?

regards

David Liddell

fieldOfView, 2022-05-06:
The official way to fix this is to wait for a new version to be released.