Ultimaker/Cura #12085: Top/Bottom

cemiyetadami, 2022-05-04:
### Application Version

5.0.0-beta+1

### Platform

Windows 10

### Printer

Ender3v2

### Reproduction steps

under top/bottom  Top/Bottom pattern and Bottom Pattern Initial Layer have no effect 
when i make selection cura forces e to slice again but no difference. allways bottom layer and initial layer in concentric pattern

### Actual results

making selection doesnt chance pattern

### Expected results

making selection must chance pattern

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

GregValiant, 2022-05-04:
@cemiyetadami Can you provide a project file (File | Save Project).  I can't duplicate this as switching back and forth works for me.

Asterchades, 2022-05-05:
It does this when Spiralize Outer Contour is enabled. Everything related to the Bottom Layer/s is ignored in 5.0b1 - you get one (1) layer with a concentric pattern for the base and that's all, as opposed to previous versions where the base would still be printed using the given settings.

This is a duplicate of https://github.com/Ultimaker/Cura/issues/12075.

closed 2022-05-23 by github-actions[bot]

PhilBaz, 2022-05-26:
Ive made additional entries in my duplicate https://github.com/Ultimaker/Cura/issues/12075 if anyone is watching this.