Ultimaker/Cura #12086: Cura 5.0.0-beta+1 GUI performance exhibits 15+ second lag

Anathema-Device, 2022-05-04:
### Application Version

5.0.0-beta+1

### Platform

Windows 11

### Printer

Custom FFFF

### Reproduction steps

Add STLs to the buildplate one at a time until the total size of the files is somewhere over 5M and the UI will become unusable.
This seems to happen on Windows 11 using a Nivida 3060 GPU.  When you select and try and move the model there will be a huge spike on the GPU then the GPU activity goes to almost nothing until the delay is past.  The behaviour limits what can be placed to only one or two models as beyond that the program is unusable.

Note: I had a friend try it on Pop-OS (an Ubuntu clone) and the problem does not exist there nor does it exist on my Surface 4 laptop with integrated Nivida graphics.

### Actual results

When there are too many (like more than 2) models on the build plate the UI is unusable

### Expected results

There should be no noticeable UI lag as the build plate is filled and the objects are arranged

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

If further diagnostic data would be useful please let me know.
[image]

Anathema-Device, 2022-05-06:
[cura.log](https://github.com/Ultimaker/Cura/files/8643655/cura.log)
Here is the cura.log file from ...\AppData\Roaming\cura\5.0 on the machine that exhibits the issue.
We also tried another machine, Windows 10 with a similar (but older) graphics card and it does not exhibit the problem so it could be either Windows 11 or possibly the newer 3D graphics driver?
Thank you for investigating this.

fieldOfView, 2022-05-06:
Could you try disabling the Sidebar GUI plugin? It has been known to cause this issue for some people before. I had hoped that the update to Qt6 would solve the issue, but perhaps it does not.

Anathema-Device, 2022-05-06:
I have previously removed all plugins and reinstalled Cura 5 and the problem still persisted plugins or no.  Right now I have the plugins that I like installed because it does not seem to impact the performance one way or another.

nallath, 2022-06-15:
This was fixed in the 5.0 final release.