Ultimaker/Cura #12087: Color scheme "Line type" -> add "Bridge"

tispokes, 2022-05-04:
### Is your feature request related to a problem?

I would like to see which line is a specific bridge, when "Enable Bridge Settings" is selected.

### Describe the solution you'd like

Additional line type "Bridge"

### Describe alternatives you've considered

n/a

### Affected users and/or printers

all

MariMakes, 2022-11-01:
Hey @tispokes,

Sorry, it took us a while to get back to you 😞 
But I saw your request and I am really intrigued. 🤩 

I'll bring it up to the team to see what we can do about it. 🤞

MariMakes, 2022-11-02:
Hey @tispokes,

I brought your issue up to the team,/ 
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-9839

Thanks for the report! 👍