Ultimaker/Cura #12089: [5.0 beta]  Reloading a project after overwriting imports models incorrectly

printingotb, 2022-05-04:
### Application Version

5.0 beta

### Platform

win10

### Printer

custom fff

### Reproduction steps

1 Load the project.
2 Save project and overwrite the same file.
3 Click reload

### Actual results

All models are now duplicates of the same model when using the reload function.

With import or drag&drop, it works as intended.

### Expected results

Should work as it does with import or drag&drop.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project:
[image]
Export and overwrite project, then hit "reload":
[image]
Result:
[image]
[OpticoolerFunels.3mf.zip](https://github.com/Ultimaker/Cura/files/8621134/OpticoolerFunels.3mf.zip)

GregValiant, 2022-05-04:
That's an odd one.
If Reload is hit right after saving then one or more models is replaced.  If the build plate is cleared after the "reload" and the file is opened in a regular way then it opens correctly.

printingotb, 2022-05-05:
Hmm, If I first clear the buildplate and then hit Reload, no objects at all appear on the buildplate. It stays clear. Nothing in the object list either.

fvrmr, 2022-05-05:
Hi @printingotb thank you for your report.
I can reproduce this so I will bring it up with the team. Keep you posted!

fvrmr, 2022-05-09:
It seems like that this behaviour is not defined. We have ticket on our backlog for this, devs see CURA-7339

MariMakes, 2024-01-31:
Quick update from our side 👋

I believe this issue is resolved with this pull request from the community: https://github.com/Ultimaker/Cura/pull/16610 which will be available with the next release. 

We recently released a Nightly version, which already includes this fix, so if you want to you can already try it out here https://github.com/Ultimaker/Cura/discussions/18080

To me, it doesn't seem like a bug in Ultimaker Cura, since we have a solution in place. Apologies for any inconvenience caused by our delayed response. I'll be closing this issue. I hope you understand.

closed 2024-01-31 by MariMakes